NatLabRockies/flasc #15: Update requirements

paulf81, 2022-08-16:
I think that the current requirements in the setup file and separate requirements doc were set by @Bartdoekemeijer to specify a not-too recent package for a few modules to avoid an issue in pandas.  I think with the most recent version of pandas we can revert to specifying a minimum version, rather than a precise version or max version.  @misi9170 would you take a look?

Bartdoekemeijer, 2022-08-19:
Hi @paulf81, you are correct! I fixed FLASC to a specific `pandas` version range,

``pandas>=1.3.0,<=1.4.0``

To avoid some issues with pandas time resampling functionalities (I think it was in `pd.downsample` or `pd.rolling`) that started appearing after version `1.4.0`. Perhaps that is solved with a more recent release of Pandas. It's worth churning through the example scripts with the latest Pandas version. If nothing comes up, I think you should be safe to move forward with the latest Pandas version! You'll still want to restrict the requirements such that at the least it skip the versions (I'm guessing 1.4.1 and possibly onward) that yield errors with the time resampling functions.

paulf81, 2022-08-19:
Thanks for the clarification @Bartdoekemeijer, I think @misi9170  has a plan and I do believe that if we use the latest versions of all these modules will avoid the issue.  Agree that specifying a version past this issue does make sense in case of installing into an existing environment

misi9170, 2022-08-22:
Thanks @Bartdoekemeijer. I haven't been able to reproduce the original bug. I do see mentions in recent pandas releases of bugs/regressions with `rolling` and `resample`; but I don't receive an obvious error when running through the raw_data_processing examples using pandas 1.4.1, 1.4.2, or 1.4.3 (latest). 

To be clear, I'm mostly focused on `a_06a...` as it calls `df_downsample`, which in turn calls `df.resample`. I actually don't see any example code that results in a call to `df.rolling`, as `df_movingaverage` does not seem to be called anywhere other than its own unit test. What am I missing here?

Bartdoekemeijer, 2022-08-23:
Hi @misi9170. Those were the examples I was thinking of too. Unfortunately I never really logged this bug properly, so it's hard to reproduce. I wish I had done this, in hindsight! I would say, if it's not giving you any issues in your work nor the example scripts,  feel free to go ahead and set `pandas` to its most recent version. It's probably fine. If it's not, we'll see soon enough!

misi9170, 2022-08-23:
Will do, thanks!

PR #19 (2022-09-02): Update package requirements
Feature or improvement description
Remove all maximum package dependencies; also remove all minimum dependencies, with the exception of:
floris>=3.1
pandas>=1.4.3

Related issue, if one exists
Closes #15 

Impacted areas of the software
`setup.py` and `requirements.txt`

Additional supporting information
None.

Test results, if applicable
None.
closed 2022-09-02 by bayc