NatLabRockies/flasc #21: [BUG] Failing tests due to pandas pd.concat

bayc, 2022-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

The test `df_time_operations.py` is failing due to type errors. The issue has been traced to a `pd.concat` call in `time_operations.py` where a column is trying to be added, but instead, rows are added with the column.

### Expected Behavior

The test should pass.

### Steps To Reproduce

Run `pytest df_time_operations.py` in the tests directory.

### Environment

```markdown
- OS: Ubuntu 20.04
- pip environment (can be retrieved with `pip list`):
  - pandas 1.4.4
  - numpy 1.21.5
  - scipy 1.8.0
```

### Anything else?

None.

misi9170, 2022-09-05:
This seems to be at least partly due to the new version of pandas. Looking into this now.

misi9170, 2022-09-05:
Have submitted pull request for review on this issue.

paulf81, 2022-10-18:
Resolved by pull request #23

closed 2022-10-18 by paulf81