NatLabRockies/flasc #27: [BUG] Dealing with turbine curtailment in FLASC

Bartdoekemeijer, 2022-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Currently, FLASC marks a turbine's individual measurements as faulty if that turbine is in curtailed operation. That ensures that all non-NaN measurements of that turbine follow the nominal power and thrust curve. However, measurements of turbines neighboring a curtailed turbine remain unaffected in the FLASC analyses. In reality, their power production can be significantly affected because curtailed turbines shed shallower wakes.

If we compare this to FLORIS model predictions which do not account for curtailment, we are not drawing fair comparisons.

### Expected Behavior

A solution to this would be to mark the measurements of _all_ turbines as faulty if at least one turbine is in curtailed operation. However, if multiple turbines regularly demonstrate curtailment, this could potentially eliminate a large share of your dataset. A less restrictive solution could be to nullify the measurements of the curtailed turbine and all turbines downstream of it.

Another solution is to create FLORIS models for each curtailment situation and generate FLORIS predictions for the SCADA in accordance with which turbines are downrated and which moments in time. This requires knowing when which turbines are in curtailed operation.

It's worthwhile thinking about the best way to approach this.

### Steps To Reproduce

N/A

### Environment

```markdown
N/A
```

paulf81, 2022-10-11:
Hi @Bartdoekemeijer, I was thinking about this and one idea I had is to add a new set of columns to the dataframe, something like pow_rated_... then under normal operations this is simply the rated power, but in curtailed operation is the value being curtailed to, or could also be a way to represent over-powering situations

We'd need to set the methods to know to use this term but could add helper functions like mark_all_curtailed_faulty or other stuff?

Bartdoekemeijer, 2022-10-12:
Hi @paulf81, thanks for sharing your thoughts. I'd like this feature to be as broad as possible: covering regular curtailment, noise-reduced operation, or even controller toggling that would yield different turbine performance curves. Including a `pow_rated` column would not allow us to distinguish different methods of curtailment. I think a column `curtailment_mode` or `operation_mode` would. For example, `operation_mode=0` would implicate regular turbine operation, and `operation_mode=1` could correspond to noise-reduced operation, and `operation_mode=2` could correspond to another type of curtailment, and so on. Each user decides for themselves/their farm what each `operation_mode` corresponds to.

Practically, here's a workflow that I think would then meet all our needs:
1. Create a FLORIS model for regular operation
2. Create as many additional FLORIS models as needed to cover each curtailment mode.
3. In our dataset, for each timestep, if we want to generate the FLORIS predictions, we would: a. Take the base FLORIS model without curtailment. b. Replace the performance curves of all turbines in `operation_mode=1` with the performance curves of the corresponding floris model. Now do the same for `operation_mode=2`, `operation_mode=3`, ... until the performance curve of each turbine matches its operation mode.
4. Evaluate the FLORIS output using `calculate_wake` and `get_turbine_power`.

This does require that users know in what operation mode each turbine is for each measurement timestamp, but I think usually that is the case. Curtailment is often scheduled according to time. If you don't know the operation mode, then the user either has to make assumptions there or simply has to mark the data is faulty, which I think is a fair decision.

What do you think @paulf81?

paulf81, 2022-10-12:
I think it's good, one thing I'm also wondering, could we not add curtailment to FLORIS as an optional input, such it could handled by passing in the derating point and clipping to there?  (I guess what value Ct should take would take a little thought) but if FLORIS had an option to accept derating modes at the interface level, then we could solve this all at once through one time-series calculate wake, but maybe I'm missing how complicated this would in fact be,

I think your approach will work, although in step 3 are we catching the impact of curtailment on wakes?  I think so but want to make sure I fully understand

We have a lot of changes in the works for the turbine.py module within FLORIS, maybe having a smart way to handle curtailment, derating and other modes within turbine.py would be an elegant general solution?

Bartdoekemeijer, 2022-10-13:
In my understanding, curtailment and really any non-regular operation mode comes down to a different power and thrust curve for the turbine. I think the way in which the performance curves change under curtailment is not trivial and can differ per operation mode, so a simple clipping would not cover all the relevant cases. Also, I don't think implementing curtailment fundamentally requires any modifications to FLORIS or the FLORIS interface.

The floris input file for a particular wind farm with 4 different operation modes across the turbines simultaneously can already be evaluated as:
```
[...]
farm:
  layout_x:
  - 0.0
  - 630.0
  - 1260.0
  - 1890.0
  layout_y:
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  turbine_type:
  - nrel_5MW
  - nrel_5MW_curtailed2300kw
  - nrel_5MW_noisereducedoperation
  - nrel_5MW_batcurtailment
[...]
```

This only requires the user to define the appropriate performance curves for each curtailment mode. I do like your idea of making it easy for users to approximate a curtailed performance curve using some rules of thumb. Perhaps this could be a preprocessor step in `floris.tools`? Something like, you insert the nominal performance curves and the curtailment limit, and out comes a curtailed turbine definition file. This would make it easy for a user to add curtailment to any turbine if they so desire.

On the FLASC end, I suggest the more generic approach where for each turbine we define an operation mode. If for a particular timestamp we would have
```
operation_mode_000=0
operation_mode_001=2
operation_mode_002=0
operation_mode_003=0
operation_mode_004=1
operation_mode_005=1
```
Then FLASC would update the turbine performance curves inside the floris object reflecting a floris input file that would look something like:
```
[...]
farm:
  layout_x:
  - 0.0
  - 600.0
  - 1200.0
  - 1800.0
  - 2400.0
  - 3000.0
  layout_y:
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  turbine_type:
  - nrel_5MW
  - nrel_5MW_operationmode_2
  - nrel_5MW
  - nrel_5MW
  - nrel_5MW_operationmode_1
  - nrel_5MW_operationmode_1
[...]
```

Now, FLASC doesn't need to 'create' an input file like above, it can modify the turbine performance curves directly using `fi.reinitialize(...)`, but you get the idea.

Also, since each curtailment mode is its own turbine definition inside floris, we have full freedom to also modify the thrust curves for each curtailment mode. So yes, we could include the impact of curtailment on the wake. In theory, I think this approach would also be able to capture axial induction control in FLORIS without making any modifications to the underlying FLORIS code.

paulf81, 2022-10-13:
Ok this is persuasive, I think this is the right approach

It seems to me like running with these options would be bound up to running in time series mode, unless any turbines are always in a particular mode, does that match where you are thinking of going with this?

Bartdoekemeijer, 2022-10-20:
> It seems to me like running with these options would be bound up to running in time series mode, unless any turbines are always in a particular mode, does that match where you are thinking of going with this?

We'd need to think about how to evaluate the curtailment scenarios with as little overhead as possible. Since different scenarios (which turbines are curtailed at any moment in time) will effectively each mean a different floris object. I guess it would be something like:
1. Identify all possible scenarios: which turbines are curtailed, when, and at the same time as which other turbines?
2. Loop through each scenario one-by-one. For each scenario, generate the correct FLORIS object and calculate the solutions.

paulf81, 2022-10-20:
OK yes that does make sense, maybe there could be some first pass like, sweep through and identify every unique combination of dertatings and curtailments, then for each case above some threshold number of points or cumulative time, develop a FLORIS model?  Then drop the remaining combinations that only happened once or something?

Bartdoekemeijer, 2022-10-31:
> OK yes that does make sense, maybe there could be some first pass like, sweep through and identify every unique combination of dertatings and curtailments, then for each case above some threshold [...]

Yes, I think something like that!

paulf81, 2023-03-23:
Is this issue now ready to close?

Bartdoekemeijer, 2023-03-24:
Yes, closing!

closed 2023-03-24 by Bartdoekemeijer