NatLabRockies/flasc #28: Change in df_movingaverage wrt to time column

paulf81, 2022-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

This may or not be what we intended so wanted to raise this as a possible issue and see if we prefer it as is or if we want to revert.  One behavior change when we updated df_movingaverage is that in the former code, the code allowed for either there was a column called 'time' and if not, then assume 'time' was the index.  Now it is required 'time' is column, which caused an error in some processes here.

[image]

The above image shows the responsible change.

I was thinking though I couldn't decide, maybe it is nice to standardize on time is a column, and any process which wants it to be an index is responsible for making a local version that way, so don't assume we wanted to change it back.  Just wanted flag for discussion.  @Bartdoekemeijer  and @misi9170  what do you think?

### Environment

```markdown
- OS:
- pip environment (can be retrieved with `pip list`):
```

Bartdoekemeijer, 2022-10-18:
@paulf81 I completely agree with you! Time must be a column by default and not the index. Also, `.ftr` files cannot hold a DataFrame in which the indices are timestamps, so default loading/writing files additionally becomes more consistent with this convention.

paulf81, 2022-10-20:
ok, in that case I think we have a few more places where we should enforce that time can't be passed in the index,

ws_pow_filtering.py: save_df()

time_operations.py: df_downsample()

floris_tools(): interpolate_floris_from_df_approx()

Each of these features a code block that does form of checking if time is a column and then setting/resetting index, or changing behavior, maybe in each of these we should adopt what we did above and insist time be a column?

Bartdoekemeijer, 2022-10-31:
> Each of these features a code block that does form of checking if time is a column and then setting/resetting index, or changing behavior, maybe in each of these we should adopt what we did above and insist time be a column?

I think that is correct! It could be something simple like:
```
if not "time" in df.columns:
    raise KeyError("No column 'time' found in your dataframe. If 'time' is the index, use 'df = df.reset_index(drop=False)' before inserting it into FLASC functions.")
```

paulf81, 2023-12-12:
@misi9170 and @paulf81 decide to archive this one and revisit later

closed 2023-12-12 by paulf81