NatLabRockies/flasc #31: Plot energy ratio gain

paulf81, 2022-10-20:


paulf81, 2022-10-20:
hi @Bartdoekemeijer, @ejsimley and @misi9170   I'm assigning this one to myself but was hoping to ask your input on getting started.  Basically, I've been looking at how we had done this previously (calculating and plotting the gain of one energy ratio to another) and was thinking the right place to work this in would be into the energy_ratio_suite framework since it only makes sense if you have two things.  In the sort of standard case you'd have a df now for scada_baseline, scada_control, floris_baseline, floris_control, and maybe we'd have a function option to plot the gains and you just have indicate which ones to make a gain on, does that make sense?

paulf81, 2022-10-20:
btw for reference I've been looking at FLORIS v2.3 which had this functionality

misi9170, 2022-10-21:
as you suggested, energy_ratio_suite looks like the place for this based on its docstring. I'll let Bart confirm though!

Bartdoekemeijer, 2022-10-31:
Yes agreed! And the indicator to decide which to make a gain on should be a list. Namely, you should specify for each case to which other case should be normalized. That'll allow you to plot scada_control normalized by scada_baseline and floris_control normalized by floris_baseline in a single function call.

misi9170, 2023-03-22:
Closing as #58 was merged.

closed 2023-03-22 by misi9170