NatLabRockies/flasc #48: [Feature]: Turn examples into a separate repository

Bartdoekemeijer, 2022-11-17:
### Description

The way we work with FLASC is that we usually create one Git repository for each wind farm that we do analysis on. Taking an example repository for the Shell-owned OWEZ wind farm, we typically have a folder structure like:

```
wfc_owez
  - tests/
      - __init__.py
      - test_wfc_owez.py
  - wfc_owez/
      - __init__.py
      - wfc_owez_cc_floris_model.yaml
      - wfc_owez_gch_floris_model.yaml
      - models.py
      - sql_interface.py
      - windrose.py
  - raw_data_processing/
      - a_00_initial_download.py
      - a_01_to_common_format_df.py
      - a_02_basic_filters.py
      - a_03_check_sensor_stuck_faults.py
      - a_04_wspowercurve_filtering.py
      - a_05a_plot_faults_with_layout.py
      - a_05b_cross_compare_wd_measurement_calibrations.py
      - a_06a_determine_timeshift_datasources.py
      - a_06b_apply_timeshift_dfs.py
      - a_07a_estimate_wd_bias_per_turbine.py
      - a_07b_wd_bias_to_df.py
      - a_08_plot_energy_ratios.py
  - energy_ratio_analysis/
      - energy_ratio_in_turbine_array.py
      - energy_ratio_vs_wd.py
      - estimate_heterogeneity_from_energy_ratios.py
  - yield_calculations/
      - calculate_aep.py
  - .gitignore
  - setup.py
  - setup.cfg
```

Behind the scenes, we have often copied over this folder structure from an existing wind farm repository to create new repositories, but this has been hidden from the public. Our current `examples` folder insufficiently represents this structure and can make it complicated for new users to build well-structured repositories and analyses.

I would like to create a template repository like this, `flasc_windfarm_example`, which contains this folder structure and readily includes the examples. This would either completely replace the examples folder in `flasc` or supplement it. When creating a new repository, one can clone this repository and adjust it accordingly for the user's wind farm. This also gives us much greater freedom in the examples that we build. Specifically, since these repositories are Python installable with `pip`, it allows us to build classes and functions inside the `models.py` library that modularly construct, for example, FlorisInterface objects with different combinations of turbine curtailment. For example, consider functions that if we plug in a timestamp it'll return the right FlorisInterface object (e.g., due to noise-reduced operation at certain times in the day). 

If we were to additionally make `flasc` available in PyPI, we can include any particular version of `flasc` in the requirements in the `setup.py` and the right version for that repo will automatically be installed for new users starting work on that repo.

@misi9170 and @paulf81, what do you think? And would this replace or supplement the current example folders?

paulf81, 2022-11-17:
I agree 100%, this would stop me from doing ad-hoc copies from existing projects and I would move to this template repo as well.  Great idea!!

paulf81, 2023-02-09:
This is excellent, thank you @Bartdoekemeijer !

Bartdoekemeijer, 2023-03-22:
I feel comfortable enough with https://github.com/NREL/flasc_cookiecutter_template to remove the `examples` folder from the `flasc:develop` branch. What do you think? If so, I can go ahead and open a PR. The PR should also include a brief description in the FLASC README file on how we use `flasc_cookiecutter_template` for our workflows.

closed 2023-07-03 by Bartdoekemeijer