NatLabRockies/flasc #49: [Feature]: FLASC on PyPI

Bartdoekemeijer, 2022-11-21:
### Description

I would like to have flasc available through PyPI such that we can include it and have it auto-installed through the `setup.py` Python module file, e.g., for #48.

Also, it would allow users to download `flasc` with a single call,
`pip install flasc`
`pip install flasc==1.0`

@rafmudaf were you the one to do this for `floris`? How complicated is this process?

Bartdoekemeijer, 2023-06-15:
It appears that #88 stems from a problem with the PyPI installation. Let's revise any release a v1.4 soon, right after we merge #86.

paulf81, 2023-07-14:
hi @Bartdoekemeijer do you know, is this issue resolved and we can close this one?

Bartdoekemeijer, 2023-07-17:
Yes, resolved! Closing.

closed 2023-07-17 by Bartdoekemeijer