NatLabRockies/flasc #55: [Feature]: Rework examples around open Smarteole database

paulf81, 2022-12-01:
### Description

A dataset from a wake steering campaign has been made publically available.  This data is provided by Engie from the Wind Farm Control campaign at Sole du Moulin Vieux and is available here:

https://zenodo.org/record/7342466

Further @ejsimley points us to code on how to download data from zenodo:

https://github.com/charlie9578/OpenOA/blob/feature/cubico-projects/examples/project_Cubico.py

Dataset should not need much filtering but can still probably show some pre-processing syntax, however, best opportunities are for showing SCADA / FLORIS analysis using real data . 

@Bartdoekemeijer wanted to flag you on this to bring

Bartdoekemeijer, 2022-12-02:
Love this idea!

paulf81, 2023-03-23:
@ejsimley is working on this one, but note this is now heading over to flasc_cookie_cutter here:

https://github.com/NREL/flasc_cookiecutter_template/issues/2

Bartdoekemeijer, 2023-03-24:
Closing this one in that case, if you don't mind

closed 2023-03-24 by Bartdoekemeijer