NatLabRockies/flasc #57: [Feature]: Indicate turbine dependencies for case to case testing

misi9170, 2022-12-13:
### Description

For large farms, we need a way of determining which (upstream) turbines a given turbine's power depends on.

Bartdoekemeijer, 2022-12-15:
You could look at

https://github.com/NREL/floris/blob/39c466000b1874e06a6f58da9c30bb877fc8d4d2/floris/tools/optimization/yaw_optimization/yaw_optimization_tools.py#L21

and

https://github.com/NREL/floris/blob/6e856e54753f290de83eeeb74b7bc97c35fe22ad/floris/tools/optimization/yaw_optimization/yaw_optimization_tools.py#L20

These functions rely on a simple geometrical wake model to quickly determine which turbines fall inside the wake of another turbine.

misi9170, 2022-12-15:
@Bartdoekemeijer thanks for pointing these out! I'll take a look before implementing anything. One 'feature' for this that Paul and I have been discussing is that we'd like to set a threshold for dependency (user defined, but default at 1% of rated power), so turbines far downstream (but still technically in the wake) of an upstream turbine are not considered dependent.

paulf81, 2022-12-15:
Thanks @Bartdoekemeijer !  These are a good pointer, but do agree with @misi9170 this new method has some novelties.  Also it's a bit inverted of these, it's making a list of upstream turbines

Bartdoekemeijer, 2022-12-16:
I see! The methods above are really simplified and do not include any kind of quantitative wake loss. You could include a downstream distance threshold here for dependency, e.g., 30 rotor diameters. I can imagine having a threshold for wake loss dependency would require more complicated methods and could therefore be a lot slower but more accurate. Depending on your use case, I can see either work!

paulf81, 2022-12-16:
I think that is a nice way to think of it, both are options, and I think the geometric approach will be much faster, but the wake model approach will allow criteria based on relative level of impact

misi9170, 2023-03-02:
Closed by PR #70

closed 2023-03-02 by misi9170