NatLabRockies/flasc #106: [Feature]: Detect and correct step changes in northing calibration

misi9170, 2023-08-10:
### Description

The northing calibration tools in FLASC currently optimize for a single wind direction bias across the entire history. It would be great to have a method for detecting when there has been a step change in the northing calibration, as can happen when a yaw encoder resets, and a time-dependent northing calibration correction. 

Steps could be:
- Detect periods of steady nothing error and step changes in northing error (possibly using a single bias for all time stamps and outlier detection tools, see #36)
- Determine biases for each identified period (using existing tools where possible, possibly by separating periods into distinct dataframes to apply the northing calibration methods)
- Recombine dataframes if necessary

misi9170, 2023-12-12:
Need to: average over noise, detect large change (possibly changes in the slope of a cumulative error from a reference?)

Steps: 
1. Apply an approach to detect step changes in northing
2. Remove the steps to get a fixed bias
3. Apply exisitng northing calibration algorithm

aclerc, 2024-08-05:
This is a feature in wind-up. It uses the `ruptures` package along with several custom functions to define an optimal northing offset history for each turbine. Happy to discuss further if it's of interest