NatLabRockies/flasc #116: [BUG] Readme is out of date

paulf81, 2023-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

README file has not been updated to match current procedures with respect to running the examples.  Need to update

### Environment

```markdown
- OS:
- pip environment (can be retrieved with `pip list`):
```

misi9170, 2023-12-19:
I'm pushing this to v2.0. v2.0 will have significant restructuring of FLASC, and it makes sense to delay building out documentation until then.

ejsimley, 2024-02-07:
I was going to add a documentation issue, but then saw this one, so I'll just add my support here! 

I noticed a lot of functions are missing docstrings throughout the code (one example is the `turbine_analysis.northing_offset` module), and some need to be updated (for example, in `ws_pw_curve_filtering.filter_by_power_curve`, the arguments `ws_deadband` and `pow_deadband` aren't described in the doctstring). 

Several of us should probably choose a few modules each and do a thorough review of the code and add missing documentation before the release.

paulf81, 2024-02-07:
> I was going to add a documentation issue, but then saw this one, so I'll just add my support here!
>
> I noticed a lot of functions are missing docstrings throughout the code (one example is the [...]

100% agree, I think @misi9170  was also asking @rafmudaf how to generate API from docstrings so we should make sure to conform with that if we are going through the effort

Bartdoekemeijer, 2024-02-08:
Good points -- and a lot of this is from my lack of discipline on documentation writing/maintenance! So please feel free to delegate some modules to me too.

paulf81, 2024-02-08:
No problem!   I think it's all learning as we go and thank you!!!

misi9170, 2024-04-10:
Although this is partially addressed in #187, but I'm going to leave this open because I think there's still more to do here (the API documentation is not yet included, although we do have an open PR for this at #180)

paulf81, 2024-11-05:
@misi9170 what do you think, in the 2.1 release we did update the docs a bit, maybe fair to close this issue and can open a new one as needed with specific documentation update requests?

paulf81, 2025-09-25:
Agree!  I'm going to close now