NatLabRockies/flasc #132: [Feature]: Add filtering based on rotor speed and blade pitch

paulf81, 2023-11-10:
### Description

A useful way to filter non-nominal behavior is to filter based on the curve of rotor-speed (x) to blade pitch (y), should work similarly to power curve filtering.

Bartdoekemeijer, 2023-11-14:
This is already possible in FLASC since the generalized filtering class implementation in #78. For example, you can do this:
```
        # Filter for blade pitching while in region II
        ws_pow_filtering.filter_by_condition(
            condition=(
                (ws_pow_filtering.df[f"blade_pitch_{ti:03d}"] > 0.0) &
                (ws_pow_filtering.df[f"ws_{ti:03d}"] > 6.0) &
                (ws_pow_filtering.df[f"ws_{ti:03d}"] < 12.0)
            ),
            label="Blade pitch above 0.0 deg while wind speed between 6 and 12 m/s",
            ti=ti,
            verbose=True,
        )
```

and the plot the results with
```
        ax = ws_pow_filtering.plot_filters_custom_scatter(
            ti=ti,
            x_col=f"ws_{ti:03d}",
            y_col=f"blade_pitch_{ti:03d}",
        )
```

paulf81, 2023-11-14:
That is cool, thank you @Bartdoekemeijer !  I was thinking though, what if we made a version that worked on a similar principle as the power curve filtering, removing outliers along the pitch dimension?

Bartdoekemeijer, 2023-11-15:
Like a curve estimation and then outlier detection? That sounds useful! I wonder if we can generalize the power curve function, because the principles are identical between the different variables (power, rotor speed, pitch).

paulf81, 2023-11-15:
I think it should be not too bad also if we use the approach of standard error from the median, binned by rotor speed and error long the pitch access?

aclerc, 2024-08-05:
wind-up has a automated process for filtering based on RPM and pitch. You don't necessarily want to just filter outliers, I find it's better to understand how the turbine controller works (in wind-up for example you need to specify if the turbine pitches to stall or feather to regulate rated power) and then adjust filtering rules based on that knowledge. Happy to discuss more if it's of interest.