NatLabRockies/flasc #133: [BUG] Polars versioning

paulf81, 2023-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Having a small issue with polars versioning where version above 0.19.5 are not able to run on my mac for reasons related to:

https://github.com/pola-rs/polars/issues/12454

And starting with 0.19.13, CI is failing for a polars-test fail, but I can't debug locally because of the earlier issue.  Would like to find a longer term solution, but for now setting the requirement to 0.19.5 and I think we can revisit in a few weeks.

### Environment

```markdown
- OS:
- pip environment (can be retrieved with `pip list`):
```

paulf81, 2023-11-27:
Maybe a way to deal with this is to make a flag to our install, such if you pip install with [mac], it installs the LTS version, and installs the normal build otherwise?  Does that sound reasonable @rafmudaf ?

rafmudaf, 2023-11-27:
That sounds like a workable approach, assuming there's a corresponding description in the installation documentation, for a developer install. However, this isn't an option for installing from PyPI.

paulf81, 2024-11-05:
Going to close this issue as FLASC has updated to polars v1 and seems to work

closed 2024-11-05 by paulf81