NatLabRockies/flasc #136: [BUG] <smarteole example notebook 06 code cell 8 wd col name>

bkramak, 2023-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

in 06 example notebook in folder examples_smarteole, cell 8

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
[~\AppData\Local\Temp\ipykernel_11804\4103650034.py](https://file+.vscode-resource.vscode-cdn.net/c%3A/flasc/flasc/examples_smarteole/~/AppData/Local/Temp/ipykernel_11804/4103650034.py) in ?()
      1 # Limit the data to this region
----> 2 df_scada = df_scada[(df_scada.wd_smarteole > (start_of_offset - 20)) &
      3                     (df_scada.wd_smarteole < (end_of_offset + 20))]

[c:\ProgramData\miniconda3\envs\flasc_tests\Lib\site-packages\pandas\core\generic.py](file:///C:/ProgramData/miniconda3/envs/flasc_tests/Lib/site-packages/pandas/core/generic.py) in ?(self, name)
   6200             and name not in self._accessors
[image]

   6201             and self._info_axis._can_hold_identifiers_and_holds_name(name)
   6202         ):
   6203             return self[name]
-> 6204         return object.__getattribute__(self, name)

AttributeError: 'DataFrame' object has no attribute 'wd_smarteole'

copilot suggests the column name might have been changed to just 'wd' from 'wd_smarteole'

### Environment

```markdown
- OS:
- pip environment (can be retrieved with `pip list`):
```

paulf81, 2023-11-27:
hi!  I am sorry for this error.  I'm actually not sure what is happening here but would like to help debug.  The column wd_smarteole should have been added by the 02 notebook.  As a first few things to try, would you mind to:
1) Re-run the notebooks in order starting from 00 (or perhaps 02 is also fine) and confirm there are no errors until 06?
2) If that is true, in the cell above the one which gives the error run this code:

```
for c in df_scada.columns:
    print(c)
```

And paste the results here?  Thank you!

misi9170, 2025-09-25:
@paulf81 I just tried deleting the downloaded data, running the 02 notebook, and running the 06 notebook and that all worked for me. I believe we can close this issue as stale or out of date, and can reopen if the issue comes up again.

paulf81, 2025-09-25:
Works for me!

closed 2025-09-25 by paulf81