NatLabRockies/flasc #159: [BUG] sqlalchemy version conflict with res-smart

aclerc, 2024-01-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

I am trying to install flasc in an existing project using poetry as described in https://python-poetry.org/docs/dependency-specification#git-dependencies. This works fine with floris but when I try to install flasc I get a conflict with another dependency. The problem is flasc depends on sqlalchemy (>=1.3,<2.0) but my project already uses a higher version of sqlalchemy (^2.0.21). I tried forking flasc and removed sqlalchemy from setup.py but that did not work.

### Expected Behavior

1. I can install flasc in another project using poetry even though that project already depends on sqlalchemy (^2.0.21)

### Steps To Reproduce

I think you could reproduce this by starting a new poetry project, adding sqlalchemy, and then install flasc

paulf81, 2024-01-24:
I think now I remember, to upgrade to from the v1 to v2 sqlalchemy I need to follow this migration pattern: https://docs.sqlalchemy.org/en/20/changelog/migration_20.html, this shouldn't be too bad, I can try to get this done soon, then I can update FLASC'S dependency to 2.0 and resolve the issue

paulf81, 2024-01-24:
@aclerc I believe this is now resolved in main and develop if you wouldn't mind to check

closed 2024-02-06 by christiannvaughn

aclerc, 2024-03-01:
Yes flasc installs for me now, thank you!