NatLabRockies/flasc #171: [BUG] Fix plot_layout in energy_ratio_het_mapper

paulf81, 2024-02-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Somewhere in the v2 conversion, maybe based on updates to package dependencies, example 05, which runs the het_mapper started throwing errors in plotting related to lin interpolant.  Logging this issue to remember to fix it before release.  Current output:

```
  File "/Users/pfleming/Projects/FLASC/flasc/examples_artificial_data/03_energy_ratio/05_estimate_heterogeneity_from_energy_ratios.py", line 88, in <module>
    hm.plot_layout(plot_background_flow=True, ylim=[0.90, 1.10], pdf_save_path=pdf_save_path)
  File "/Users/pfleming/Projects/FLASC/flasc/flasc/analysis/energy_ratio_heterogeneity_mapper.py", line 323, in plot_layout
    lin_interpolant = LinearNDInterpolator(
  File "interpnd.pyx", line 290, in scipy.interpolate.interpnd.LinearNDInterpolator.__init__
  File "_qhull.pyx", line 1827, in scipy.spatial._qhull.Delaunay.__init__
  File "_qhull.pyx", line 353, in scipy.spatial._qhull._Qhull.__init__
scipy.spatial._qhull.QhullError: QH6154 Qhull precision error: Initial simplex is flat (facet 1 is coplanar with the interior point)

While executing:  | qhull d Qz Qc Q12 Qt Qbb
Options selected for Qhull 2019.1.r 2019/06/21:
  run-id 2042442214  delaunay  Qz-infinity-point  Qcoplanar-keep
  Q12-allow-wide  Qtriangulate  Qbbound-last  _pre-merge  _zero-centrum
  Qinterior-keep  Pgood  _max-width 6.2e+03  Error-roundoff 6.6e-12
  _one-merge 4.6e-11  Visible-distance 1.3e-11  U-max-coplanar 1.3e-11
  Width-outside 2.6e-11  _wide-facet 7.9e-11  _maxoutside 5.3e-11

The input to qhull appears to be less than 3 dimensional, or a
computation has overflowed.

Qhull could not construct a clearly convex simplex from points:
- p69(v4): 4e+02 7.1e+02     0
- p30(v3): 2.9e+03 -7.1e+02 1.4e+03
- p0(v2): 4.8e+03 -1.8e+03 4.3e+03
- p99(v1): -1.5e+03 1.8e+03 8.3e+02

The center point is coplanar with a facet, or a vertex is coplanar
with a neighboring facet.  The maximum round off error for
computing distances is 6.6e-12.  The center point, facets and distances
to the center point are as follows:

center point     1630        0     1628

facet p30 p0 p99 distance= 6.1e-15
facet p69 p0 p99 distance=    0
facet p69 p30 p99 distance= 2.8e-14
facet p69 p30 p0 distance= 2.3e-13

These points either have a maximum or minimum x-coordinate, or
they maximize the determinant for k coordinates.  Trial points
are first selected from points that maximize a coordinate.

The min and max coordinates for each dimension are:
  0:     -1494      4755  difference= 6249
  1:     -1804      1804  difference= 3608
  2:         0      4755  difference= 4755

If the input should be full dimensional, you have several options that
may determine an initial simplex:
  - use 'QJ'  to joggle the input and make it full dimensional
  - use 'QbB' to scale the points to the unit cube
  - use 'QR0' to randomly rotate the input for different maximum points
  - use 'Qs'  to search all points for the initial simplex
  - use 'En'  to specify a maximum roundoff error less than 6.6e-12.
  - trace execution with 'T3' to see the determinant for each point.

If the input is lower dimensional:
  - use 'QJ' to joggle the input and make it full dimensional
  - use 'Qbk:0Bk:0' to delete coordinate k from the input.  You should
    pick the coordinate with the least range.  The hull will have the
    correct topology.
  - determine the flat containing the points, rotate the points
    into a coordinate plane, and delete the other coordinates.
  - add one or more points to make the input full dimensional.
```

### Steps To Reproduce

Running: examples_artificial_data/03_energy_ratio/05_estimate_heterogeneity_from_energy_ratios.py

### Environment

```markdown
- OS:
- pip environment (can be retrieved with `pip list`):
```

misi9170, 2024-04-10:
@paulf81 we still need to do this

misi9170, 2024-04-10:
Closed by #189

closed 2024-04-10 by misi9170