NatLabRockies/flasc #194: [BUG] There are no PyPI releases for FLASC 1.4 and 1.5

Bartdoekemeijer, 2024-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

The only available FLASC releases on PyPI are 1.3.1 and 2.0, see https://pypi.org/project/flasc/#history.

### Expected Behavior

I would like to be able to access the latest FLASC 1.5.1 version through PyPI. The majority of my workflow still depends on this release.

### Steps To Reproduce

1. Create a new virtual environment.
2. Try to install FLASC 1.5 using `pip install flasc==1.5`

### Environment

```markdown
N/A
```

misi9170, 2024-05-17:
Hi @Bartdoekemeijer ,

There was a problem with the Package Upload action, see #190 , that we fixed with a series of commits to the develop branch and then merged onto main in #192. Part of that series of commits was to allow us to manually trigger the upload for v2.0, and then remove the manual trigger again to avoid doing it accidentally in future.

However, we did not go back and upload v1.4 and v1.5 to PyPI, and I'm actually not 100% sure how to do this (but there's bound to be a way). Howe important do you feel it is to have v1.4 and v1.5 on PyPI, vs cloning FLASC, checking out the desired version, and pip installing locally? I can see that this would be a bit of a nuisance for users that have tools built around version 1.x, as you do.

paulf81, 2024-05-17:
Hi @Bartdoekemeijer , I just tried a procedure to upload v1.4 and v1.5 (technically with some additional versioning to include the fixes to make pypi work)

[image]

Can you see if it works now?

misi9170, 2024-05-17:
@paulf81 that worked for me, as long as I also specified the patch number, i.e. these work:
`pip install flasc==1.4.2`
`pip install flasc==1.3.1`
`pip install flasc==1.5.1`
`pip install flasc==1.3`
`pip install flasc==2.0`

but these don't:
`pip install flasc==1.5`
`pip install flasc==1.4`

Still, I think that's OK in terms of addressing the issue here?

paulf81, 2024-05-17:
Yeah, I think we hit a philosophical dillema if we try to have flasc 1.3 or 1.4 uploaded since they require a bugfix to upload, they aren't exactly 1.3 or 1.4 anymore.  Maybe we can mask v1.3 in pip -> 1.3.1 since it just contains fixes to the README

misi9170, 2024-05-20:
I think it's fine to just have 1.4.2 and 1.5.1 up. If you try to `pip install flasc==1.5`, pip gives you a list of the available versions (showing 1.5.1), so I don't think anyone is going to be too confused

paulf81, 2024-05-20:
@Bartdoekemeijer good to close this one?

Bartdoekemeijer, 2024-05-21:
Awesome @misi9170! Yes perfect, thank you!

closed 2024-05-21 by Bartdoekemeijer