NatLabRockies/flasc #198: Tighten up nomenclature

paulf81, 2024-07-24:
### Description

In the process of completing #197 , I notice that there's an opportunity to standardize the terms for things in FLASC, especially for functions that were written fairly spaced apart in time. 

This also could relate to upcoming work on implementing yaw-change detection (can link as that comes online).

 For now just opening this issue to track some things we might focus on.  Maybe could ulimately open nomenclature page in documetation and nail this down?  Start with a discussion?  What do you think @ejsimley  @misi9170 ?

turb_id/turb_no/turbid - Should probably just pick one of these
Yaw/yaw_offset/etc.

Bartdoekemeijer, 2024-08-05:
FYI, happy to help populate this list!

paulf81, 2024-08-09:
@Bartdoekemeijer  and @misi9170 I opened #201 as a place to fill out this list