NatLabRockies/flasc #224: [Feature]: Update panda to copy-on-write and re-enable warnings

paulf81, 2024-10-18:
### Description

FLASC makes extensive use of pandas.  Currently many of the scripts in flasc use warning suppressions, and many of these are warning about changes made to copies to dataframes.  Beginning as an option in pandas 2 and becoming default in pandas 3, is the option to enable a [copy-on-write](https://pandas.pydata.org/docs/user_guide/copy_on_write.html#copy-on-write-chained-assignment) behavior.  Enabling this throughout the flasc should:

- Lower risk of bugs
- Gain performance improvements
- Remove the need to suppress warnings

Propose we should work this in soon

Bartdoekemeijer, 2024-10-19:
Sounds great! This always felt like a big challenge working with Pandas Dataframes.

misi9170, 2025-10-07:
Good question---I'm not sure. Although I addressed the deprecation warnings that were being printed in examples in #253, I didn't try to undo any existing warning suppression in the examples and I have not gone through and tried to thoroughly ensure that pandas is copying. Would be good to take another pass through the examples and see if/where we have warning suppression and if any other warnings arise. Among other things, I mostly took care of `DeprecationWarning`s, but there may still be some dreaded "copy of a slice" warnings