NatLabRockies/flasc #252: [BUG] DeprecationWarning

maapasan, 2025-08-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Deprecation warnings appear in several functions, due to a change of argument names in some libraries.
May give errors in the future

- [ ] [flasc/data_processing/energy_ratio_wd_bias_estimation.py:L118](https://github.com/NREL/flasc/blob/cab01a4d76d970435aceb1e5af3c75efd9cccbf5/flasc/data_processing/energy_ratio_wd_bias_estimation.py#L118) 
DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)
    self.logger.info("  Constructing energy table for wd_bias of %.2f deg." % wd_bias)

- [ ] [flasc/data_processing/energy_ratio_wd_bias_estimation.py:237](https://github.com/NREL/flasc/blob/cab01a4d76d970435aceb1e5af3c75efd9cccbf5/flasc/data_processing/energy_ratio_wd_bias_estimation.py#L235) 
DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)
    + " WD bias: %.3f deg." % wd_bias

- [ ] [flasc/utilities/energy_ratio_utilities.py:580](https://github.com/NREL/flasc/blob/cab01a4d76d970435aceb1e5af3c75efd9cccbf5/flasc/utilities/energy_ratio_utilities.py#L580): 
DeprecationWarning: `pl.count()` is deprecated. Please use `pl.len()` instead.
  (Deprecated in version 0.20.5)
    .agg([pl.mean("pow_ref"), pl.mean("pow_test"), pl.count()])

- [ ] [flasc/utilities/energy_ratio_utilities.py:582](https://github.com/NREL/flasc/blob/cab01a4d76d970435aceb1e5af3c75efd9cccbf5/flasc/utilities/energy_ratio_utilities.py#L582): 
DeprecationWarning: `pl.count()` is deprecated. Please use `pl.len()` instead.
  (Deprecated in version 0.20.5)
    .filter(pl.count().over(bin_cols_without_df_name) == num_df)

- [ ] [flasc/analysis/energy_ratio.py:125](https://github.com/NREL/flasc/blob/cab01a4d76d970435aceb1e5af3c75efd9cccbf5/flasc/analysis/energy_ratio.py#L125): 
DeprecationWarning: the argument `columns` for `DataFrame.pivot` is deprecated. It was renamed to `on` in version 1.0.0.
    df_.with_columns(

### Environment

```markdown
- OS:
- pip environment (can be retrieved with `pip list`):
```

pablo-benito, 2025-08-01:
sorry, we should have mentioned that we hit those Deprecation Warnings using flasc under _pytest_. Even on your own github tests, you get all those Deprecation Warnings (and more):

Look:
https://github.com/NREL/flasc/actions/runs/16681959638/job/47222510974#step:6:111

As you can see, pytest raise all those warnings that we mentioned. But there are some others, including some flasc Deprecation warnings (!)

misi9170, 2025-09-25:
Addressed in #253